Duke-BME-Design/Ureter #1: Discuss 2 Wheatstone Bridge Circuit

kmc97, 2018-01-12:


yashasaxena, 2018-01-17:
Hi Dr. Palmeri @mlp6 ,

After our final presentation, you made a suggestion about coupling our wheatstone bridge circuit with another wheatstone bridge that could cancel out the noise coming from the catheter with readings from another pressure catheter? After three weeks, we kind of forgot the specifics of your suggestion and were hoping you could clarify! Deepti and I will be working on denoising and amplifying the signal for the next week or so, so would be great to get your two cents. I attached images of our circuit and initial Pressure-Voltage readings. We have not connected the low pass filter yet. 

[image]

[image]

mlp6, 2018-01-22:
Hmmmm... I remember us briefly discussing this during your presentation, but I have forgotten exactly what I said / was referring to.  I think that instead of having a second Wheatstone bridge, you want to try to introduce some noise-cancelling elements into your existing bridge.  We can discuss during our scrum this week.

yashasaxena, 2018-01-29:
discussed

closed 2018-01-29 by yashasaxena