Duke-BME-Design/Ureter #6: Observe Signal to Noise Reading

kmc97, 2018-01-22:
Figure out the basis of the Signal to Noise ratio. A baseline to improve

kmc97, 2018-01-26:
delayed due to reconstruction of circuit. in process of protoboarding to reduce some of the bouncy signal

kmc97, 2018-02-08:
now that pressure circuit is calibrated, will move on to this next

kmc97, 2018-02-09:
tried to institute 60 hz notch filter, took away noise but voltage response is MAJOR delayed.. honestly doesnt fluctuate that significantly-- may be ok
[image]

kmc97, 2018-02-09:
if SNR is calculated by the mean/st.dev then the circuit has a SNR of 4.5 and the catheter has a SNR or 42... which seems sort of incorrect

yashasaxena, 2018-02-12:
We could use FT to calculate SNR? I think in the time domain we would want to have the clean signal compared to the dirty signal minus the clean signal

yashasaxena, 2018-02-12:
We might also try getting the reading at 0 pressure and considering that to be just the noise?

closed 2018-04-21 by deeptig94