Duke-BME-Design/Ureter #14: Pressure Testing Environment with Delta 5 mm HG

yashasaxena, 2018-01-29:
Develop a simple pressure testing environment to determine how our circuit responds to change

yashasaxena, 2018-01-29:
maybe? https://jlbrooks.co.uk/media/catalog/product/cache/1/image/800x800/9df78eab33525d08d6e5fb8d27136e95/d/-/d-capri-ora200.jpg

yashasaxena, 2018-01-31:
https://www.amazon.com/dp/B01MCZ646P/ref=sspa_dk_detail_0?pd_rd_i=B010IS0OUE&pd_rd_wg=OSrkJ&pd_rd_r=7CQC8KMNRDV56MWWC14F&pd_rd_w=dUtea&th=1

yashasaxena, 2018-02-03:
Going to be a U-tube manometer with one end capped off with a pressure modulating piston, design included in lab notebook, we can use the piston to mimic peristalsis pressure wave-form as well

yashasaxena, 2018-02-24:
had some downtime while working at the co-lab, testing out this print 
[image]

kmc97, 2018-02-24:
how will this work? is this just holding the tube in place?

yashasaxena, 2018-02-25:
ya, like a way to prop it vertical so we can assess height difference

closed 2018-04-21 by deeptig94