Duke-BME-Design/Ureter #15: Develop Test Plan for Pressure Testing

yashasaxena, 2018-01-29:
make plan for pressure testing to validate our circuit as ready

yashasaxena, 2018-01-30:
some ideas: 
- scenarios for testing (Dr. P's suggestion -> randomizing pressure val order, high to low, low to high, introduction of physiological factors - peristalsis, saline) 
- analysis of timing...capacitor induced (how long does it take for pressure values to be registered?),
- signal-to-noise ratio (set an ideal value given the magnitudes of intended readings and compare)

kmc97, 2018-02-08:
see excel sheet... there definitely is some aspect of timing delay. visually it appears to be on the order of tens of millivolts and is corrected within about 3 seconds (estimation)

yashasaxena, 2018-02-09:
ok great, i'll work on this tonight

yashasaxena, 2018-02-12:
Located in google drive: 
Team Ureter > Spring 2018 > 2/11 Test Plan

closed 2018-04-21 by deeptig94