Duke-BME-Design/Ureter #20: Draw out and label Pressure Catheter Circuit

kmc97, 2018-02-12:
In order to figure out why the circuit is not behaving as we would assume, diagram and reassess the catheter

kmc97, 2018-02-14:
have re done the circuit, the circuit diagram is at Chesterfield, we should add "expected values" to the sheet so that when we are problem solving it will be easy to check at each stage

kmc97, 2018-02-16:
it is clear that the catheter is not working ideally. it takes about 20 seconds to initialize the circuit (so that voltage values arent floating everywhere) and then the circuit is powered for 5-10 minutes, all values increase to the upper rail and the circuit must be turned off and on again.  It seems as if the circuit behaves in a differently each time we run a test-- which makes it very frustrating.  Despite this, we have refixed the circuit and have established data shown below:
[image]

In this picture we recorded a voltage response while slowly increasing the pressure.  Then we turned off the circuit, let it sit, then turned it back on, waited 30 seconds and measured again.

kmc97, 2018-02-16:
anyways-- we are going to give it another go tomorrow.  Unsure if either of you would have a suggestion for attempting de-noise this output in terms of filtering on the hardware side or software side. @mlp6 @jl532 ...

we are semi tempted to drop the catheter and move on with a pressure transducer circuit-- bc we are spending a lot of time on this and i was able to create that independent pressure circuit very cleanly and easily

mlp6, 2018-02-16:
Chances are that something associated with the catheter circuit is capacitive and accumulating charge while it remains on for a while... no discharge / reset abilities "out of the box", right?  Agree that you don't want to keep spinning your wheels... just make sure that you convince yourself this isn't a "solvable" problem.

kmc97, 2018-02-19:
our friend catheter keeps leading us on... have some questions about our circuit that i will bring up in our meeting tomorrow. have established offsets with voltage dividers that do not need to be continuously updated with each power on/off of catheter-- which is slightly promising

kmc97, 2018-02-20:
[image]
check it out @yashasaxena @mlp6

kmc97, 2018-02-20:
no longer changing filters and such of circuit... looking to develop active offset loop--- will create separate issue on git hub. also will start moving average

yashasaxena, 2018-02-21:
Nice!

closed 2018-03-24 by kmc97