Duke-BME-Design/Ureter #21: Establish Formal Quanitative Analysis for Pressure Circuit

kmc97, 2018-02-12:
- Equation to relate Pressure to Voltage
- SNR

kmc97, 2018-02-24:
[catheter_voltage_response.xlsx](https://github.com/Duke-BME-Design/Ureter/files/1754740/catheter_voltage_response.xlsx)
[cath_steady_pressure_movingAVG.xlsx](https://github.com/Duke-BME-Design/Ureter/files/1754741/cath_steady_pressure_movingAVG.xlsx)

This is initial data, implementing the moving average filter... It looks like the catheter does a good job of picking up low to high pressure changes, but evaluates high to low pressure differences in a different manner...

deeptig94, 2018-02-24:
The number in parentheses is the number of values taken by the moving average filter.

closed 2018-04-21 by deeptig94