Duke-BME-Design/Ureter #23: Stand for U-tube Manometer

yashasaxena, 2018-02-17:
Something to bolster the u-tube when we modulate pressures, set location based known pressure values 
https://www.thingiverse.com/thing:1205032

yashasaxena, 2018-02-17:
Piston that will be modulating pressure in U-tube manometer
[image]

yashasaxena, 2018-02-28:
Making brackets for this, but other than that it should be good to go for our dynamic range
[image]
[image]

closed 2018-02-28 by yashasaxena

kmc97, 2018-03-05:
that looks good! like the duck tape lol