Duke-BME-Design/Ureter #24: Optical Encoder System

yashasaxena, 2018-02-19:
start thinking about optical encoder to track catheter distance; talk to Matt Brown about existing; think about how we will attach to catheter

yashasaxena, 2018-02-21:
@mlp6 did you say we could trigger the encoder as on/off, so we could control when it starts and stops taking measurements?

mlp6, 2018-02-21:
Yes... but that is done more by the entity controlling the encoder than the encoder itself.  I can send you some details on the encoder we use in the lab if you would like that as a starting point.

yashasaxena, 2018-02-21:
@mlp6 yes please! yeah i guess it might be more when the microcontroller decides to collect whatever data is currently on the device through the analog output, and then we can always use the difference of the data first and last collected as the length measurement

yashasaxena, 2018-02-22:
@mlp6 matt says we will have to order our own as there are no free ones: https://www.usdigital.com/products/encoders/incremental/linear/LIN 
with the encoder at the bottom of the page

how do we go about ordering? also, would you say this is something we can incorporate with the cystoscope somehow?

yashasaxena, 2018-02-26:
So, I just met with Dr. Sharifi, and he does not need the measurement process to be automated. He only wants an auditory signal, so the surgeon can mark the catheter himself physically. Will open another issue for that

mlp6, 2018-02-28:
https://www.digikey.com/en/product-highlight/b/broadcom-semi/aedr-871x-digital-aedr-872x-analog-encoder-devices?utm_adgroup=xGeneral&gclid=Cj0KCQiAw9nUBRCTARIsAG11eiftp3r_CSIzVyAXj3QGxNrjH_8n7sbPS70rUZKSddcICkXr3CwJlhsaAsi0EALw_wcB

mlp6, 2018-02-28:
I am going to upload some of our associated code to GitHub (just need to check if I can keep it public or need to make it private).

mlp6, 2018-02-28:
You should all have access now to: https://github.com/mlp6/prostate_probe_rotation  The specific code of interest is probably `src/encoder.py`.

yashasaxena, 2018-02-28:
THANKS! So, in terms of getting the encoder, should I put in a formal request to either you or Matt?

mlp6, 2018-03-01:
Matt can take care of this for you.

kmc97, 2018-03-05:
ordered.

closed 2018-03-05 by kmc97