Duke-BME-Design/Ureter #29: Catheter Best Practices

yashasaxena, 2018-03-01:
Per our discussion with Matt/Andy from Millar yesterday, some things we can try to test what might be causing drift: 
- allowing catheter to power up for 30 minutes before taking readings
- trying to supply catheter with less power than 5V (internal circuit might be heating up)
- observing catheter in water vs air (more stable)

We should also record the drift we are observing over time. 
When does it start to drift? What is the relationship with drift and time? Magnitude of drift?

kmc97, 2018-03-05:
i can head over to chesty on Tuesday to try some of this--- we can talk tomorrow

deeptig94, 2018-03-05:
He also mentioned that we should try letting the catheter power up for 30 minutes in water at a temperature of 37 degrees celsius (body temperature) and also take readings while it's in water at the same temperature. Since the catheter was optimized to be used in the body and is very sensitive to temperature changes, it's functionality may improve when the temperature is kept at this constant.

yashasaxena, 2018-03-05:
nice, thanks for rememberering that one!

yashasaxena, 2018-03-10:
Deepti and I put it in a water bottle filled with RT water. 

Readings at first glance look A LOT more stable. 
- Voltage change from inserting catheter deeper into the water shows a lot greater difference
- Water seems to be a steadier environment because we don't see as many fluctuations
- Plot of inserting catheter in U-tube manometer from a 0 mm Hg point to a 5 mm Hg point:
[image]
Notes about chart: 
(after 25 secs the catheter was kept constant at 5 mm Hg)
x-axis: time in seconds
y-axis: voltage reading from arduino in Volts
Jumps are associated with Deepti moving the cather over a delta of 1 mm Hg and pausing
After 2 mm Hg, there was a 1 V jump. The lower pressure values have a much more gradual increase as seen from 0 to 10 seconds.

yashasaxena, 2018-03-10:
One more thing -
we are now powering catheter with 2.5 V instead of 5V. This may have been contributed to reduction in drift. It definitely didn't hurt to lower the power.

closed 2018-03-24 by kmc97