Duke-BME-Design/Ureter #30: set up rasberry pi

kmc97, 2018-03-05:
get it up and running

kmc97, 2018-03-06:
looks like were going to have to figure out a way how to read in analog signals with an MCP3008 thing or some discharge circuit 

https://www.allaboutcircuits.com/projects/building-raspberry-pi-controllers-part-5-reading-analog-data-with-an-rpi/

yashasaxena, 2018-03-28:
This hasn't happened yet because we're having monitor hook up issues, so its currently blocked

yashasaxena, 2018-03-30:
Monitor hook up issues resolved, now we can move forward by setting up ADC chip, and uploading code

yashasaxena, 2018-04-03:
The MCP3008 chip is reading voltage values correctly, tested by reading outputs of a high frequency 1kHZ sawtooth 0-1V wave, voltage values look right 
[image]

yashasaxena, 2018-04-03:
we have and sd card,  the tv and the long hdmi cable so we can wire out from our station to the monitor, github repo is on there now so we can push and pull code too

closed 2018-04-03 by yashasaxena