Duke-BME-Design/Ureter #33: Raspberry Pi Analog Inputs

yashasaxena, 2018-03-15:
Hi Dr. Palmeri, 

I'm starting to set up the code for the RPi. Upon first glance, I can't find a good way with just the Rpi board to read Analog inputs. 

What are your recommendations of:
- using an ADC chip: https://www.adafruit.com/product/856
- having some sort of a comparator on our board to just excite a digital pin on the Pi at the right difference (don't know how reliable this would be cus our catheter still has a lot of drift)
- sticking with the Arduino

mlp6, 2018-03-16:
Matt has a "go to" ADC for the Pi... have you asked him?

yashasaxena, 2018-03-16:
Oh oops no. I'll email him and tell the others too next week @deeptig94 @kmc97

jl532, 2018-03-19:
I have an ADC board that connects to RPi via I2C, let me dig one out if you haven't yet talked to Matt yet!

yashasaxena, 2018-03-19:
Thanks Jason! He ordered one, but in the meanwhile if we could use that it would be great!

yashasaxena, 2018-03-28:
we have the chip now

closed 2018-03-28 by yashasaxena