Duke-BME-Design/Ureter #40: Python Code for Raspberry Pi

yashasaxena, 2018-04-02:
Have some code uploaded in "rPi_codeBase" branch
need to QC once we finish setting up Pi 

should plug in analog signal from catheter into Channel 0 of ADC chip

yashasaxena, 2018-04-03:
analog values are being read properly, but i was getting all zeros for my moving average so i need to double check what's going on there

kmc97, 2018-04-10:
reading encoder position and voltages-- code pushed to branch

kmc97, 2018-04-13:
sos @mlp6 @jl532 @yashasaxena @deeptig94 cant push the code from the pi to git.... or pull for that matter-- error looks like this 
[image]

kmc97, 2018-04-13:
im gonna retype and save to my local computer just in case the pi doesn't save the data.... anyways, i have the catheter voltages being exported into a csv file-- so we can use this to graph since matplotlib hates me and my poor computer

kmc97, 2018-04-16:
going to close this and make more specific tasks

closed 2018-04-16 by kmc97