Duke-BME-Design/Ureter #51: Encoder detection validation

kmc97, 2018-04-16:
Record voltages and UVJ/orifice output to verify and validate that the rising/falling edges correspond to the issuance of a declared encoder position. this can be done by exporting csv and observing if the UVJ/orifice output changes when a voltage threshold is switched

kmc97, 2018-04-17:
[image]

kmc97, 2018-04-17:
as you can see, the blue line (UVJ-rising edge) changes at the rising edge of the voltage pulse-- the encoder correctly identifies this edge. the orange line (orifice) changes at the falling edge.  This confirms that the algorithm to detected relative encoder position is VERIFIED

kmc97, 2018-04-17:
thoughts @yashasaxena @deeptig94 ?

yashasaxena, 2018-04-18:
Yeah that looks right on the money.

closed 2018-04-21 by deeptig94